MicrosoftDocs/windows-iotcore-docs #1152: How to copy visual studio debugger to windows 10 iot core os

logeshpalani98, 2019-03-18:
Unable to Deploy Application visual studio to raspberry pi 3
 Error Message :  `DEP6957: Failed to connect to device '169.254.213.214' using Universal Authentication. Please verify the correct remote authentication mode is specified in the project debug settings. COMException - Exception from HRESULT: 0x801C0003 [0x801C0003] IoTServer`

i have deployed application from IoT Dashboard success fully but tried from visual above exception is shown,,
 i searched many sites finally got this message
> F5 driver deployment from Visual Studio does not work on Windows 10 IoT Core. Drivers must be manually copied and registered on the device

How can copy debugger manually to windows 10 IoT Core

paulmon, 2019-03-18:
Are debugging a driver?
If you are debugging a UWP application F5 should just work.
If you are debugging a console application see these steps: https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/remotedebugging

logeshpalani98, 2019-03-19:
@paulmon in Microsoft Docs gives the solution for Debugger is 

> F5 driver deployment from Visual Studio does not work on Windows 10 IoT Core. Drivers must be manually copied and registered on the device

So, I need to Copy manually (Visual Studio Debugger).

paulmon, 2019-03-19:
If you building device drivers see this page:
https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/driverdeployment

If you are developing a UWP application see link:
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/buildingappsforiotcore

If you are developing a console app see this page:
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/remotedebugging

closed 2019-03-21 by saraclay