MicrosoftDocs/windows-iotcore-docs #1172: WiFi and BT on 3B+

PeterWone, 2019-03-26:
We know 17661 is the Pi 3B+ build.

Do you have any interest whatsoever in supporting the 3B+ in any build other than 17661?

If the answer is no then say so.

Gorgonbert, 2019-04-09:
I would also like to know what the plans are for supporting RPI 3B+ 

I just went through an awful experience installing windows iot on one of those boards.

So Microsoft really thinks this is a good experience? 
1. Downloading an ISO hidden away on some page instead of simply in the dropdown on the IoT dashboard. (Why is it not shown there?)
2. You can't just write the ISO to the SD card... no... you have mount it to find an MSI
3. You have to actually install the MSI to your machine just to fish out the FFU file from a directory that's actually named "RaspberryPi2" even though it works ONLY on RPI3B+ (are you kidding me?)
4. You can finally use the IoT dashboard to write the image to the SD card.
Seriously?

PeterWone, 2019-04-09:
@Gorgonbert As it happens my very simple use of the Pi can also be done using JavaScript in Chromium on the default flavour of Linux shipped with the Pi -- which has drivers for everything including WiFi and Bluetooth, and zero installation hassles. Even the barcode scanner involved works without installing drivers. The hardest part was finding out how to boot in kiosk mode, but this is apparently a common requirement and there are instructions on the net that work.

I got lucky; more complex requirements might not be trivially done with a browser and some script. But dotnet core also works on this platform. You can even remote debug it. Some things you can't do, but not much that you ought to be doing on an embedded device.

So I'm starting to wonder about the value proposition of WinIOT on the Pi.

saraclay, 2019-05-14:
There will be another 3B+ release, but we do not have a ETA for it at the moment. We recommend using the 3B for now.

closed 2019-05-14 by saraclay

PeterWone, 2019-05-15:
@saraclay Thank you for responding with a definite answer. If I sound distrustful it's because the last two times Microsoft went silent on a technology I was trying to use were Silverlight and Windows Phone, and I am inclined to interpret extended silence as abandonment.