MicrosoftDocs/windows-iotcore-docs #1178: Iot Dashboard proxy authentication settings

trivalik, 2019-03-29:
I try to install a SD card image of build 17763 for raspberry but it comes up that there is no way to set an authentication or use the windows own (domain user is logged in). I got: 407 Proxy Authentication Required

Alternative way would be to use the proxy without authentication but there I got DLG_FLAG_SEC_CERT_CN_INVALID in Edge browser access, because the firewall use another certificate. I checked already that all certificates in "Trusted Root Certification Authorities”. The  internet explorer request is still blocked by our firewall, somehow.

What websites are getting access by IoT Dashboard, to add them into firewall exception?

saraclay, 2019-04-01:
@trivalik I was not able to repro this issue. Could you submit this issue to our Feedback Hub, using the instructions [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT)? Thank you.

closed 2019-04-01 by saraclay

trivalik, 2019-04-03:
I am not able to send this to feedback hub, since its blocked here.

Can you provide the accessed pages by IoTDashboard that we can add it to our firewall.