MicrosoftDocs/windows-iotcore-docs #1179: Iot Dashboard does not write the SD Card, no error

trivalik, 2019-03-29:
I tried at home to write and SD card with build 17763 for raspberry. The download was done but the SD card seems not updated, The IoT Dashboard shows still the old build 15xxx. I tried then another smaller 4GB sd card with no image on it, and here I could see that no write happened.

Is there any cause why this could happen without any error?

The 8GB with 15xxx image is a SanDisk EDGE 8GB HC I

saraclay, 2019-04-01:
@trivalik While I don't doubt the error you came across, I cannot reproduce this error as the dashboard offers the 17763 build. It's better to use a 16GB SD card if you can. It's possible that your SD card could not fit the recent build.

trivalik, 2019-04-03:
After looking for a cause it is maybe that drive C:\ was near full. Some software eaten up around 20 GB and I didn't got that., maybe the last VS2017 update. 
I guess that the IoT Dashboard is installing the flash.ffu on C:\Program Files\... and that file is not there, like when I manual install the msi file with the flash.ffu image.

saraclay, 2019-04-04:
That would make sense. If you clear the SD card and it still doesn't work, please feel free to create a new issue.

closed 2019-04-04 by saraclay