MicrosoftDocs/windows-iotcore-docs #1182: Minnowboard Turbot

druske, 2019-03-30:
I'd like to see the page updated to note mappings for the newer Minnowboard Turbot hardware. If there are no differences, a note to that effect would suffice. Have a close look at GPIO 9 on pin 26, though; by the schematics, the Max routed this to ball BH12 on its SoC, on Turbot this routes to BH8; a test program meant to toggle pin 26 as an output on the Turbot as GPIO pin 9 does not appear to work (the same program works fine for GPIO 0 or 1, however, on pins 21 and 23). It would be useful to understand how the logical GPIO pins are mapped to the physical balls on the SoC, and this page should at least point to such a resource.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c4b5c424-b310-c7d0-ca35-bc7582ea62d9
* Version Independent ID: fa229f25-c4ba-6e9c-fde0-fb9160661916
* Content: [Minnowboard Max Pin Mappings - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/pinmappings/pinmappingsmbm#feedback)
* Content Source: [windows-iotcore/learn-about-hardware/PinMappings/PinMappingsMBM.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/PinMappings/PinMappingsMBM.md)
* Product: **windows-iot**
* Technology: **iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-04-04:
Hi @druske, I've added a note to the Minnowboard Turbot documentation to the Pin Mappings doc.

saraclay, 2019-04-04:
@druske I've put a note to the Minnowboard Turbot documentation to the Pin Mappings doc.

closed 2019-04-04 by saraclay