MicrosoftDocs/windows-iotcore-docs #1213: Incorrect list index and Powshell script format

v-liujxu, 2019-04-08:
In **Generate Lockdown Packages** section, the index of 'Execute the following commands to generate required packages:' should be 7 in the numbered list. And more, the format powershell script is incorrect.

saraclay, 2019-04-08:
@v-liujxu do you mind including the link to the documentation you're referring to? Thank you!

v-liujxu, 2019-04-09:
@saraclay , here is the document link.
[https://docs.microsoft.com/en-us/windows/iot-core/secure-your-device/securebootandbitlocker](https://docs.microsoft.com/en-us/windows/iot-core/secure-your-device/securebootandbitlocker)

saraclay, 2019-04-11:
Thank you. We're in the process of updating this doc, as well as a few others. We'll see if this issue is fixed in the update.

closed 2019-05-16 by saraclay