MicrosoftDocs/windows-iotcore-docs #1243: Additional Provisioning Details Needed

druske, 2019-04-23:
I have two systems set up with VS2017 and the latest SDK and WDK released for Windows 1809 (10.0.17763). Contrary to instructions, VS2017 does not prompt for Device Type on the Device Configuration dialog. I can deploy regular apps to a Minnowboard Turbot just fine, and ping both ways just fine, and I can build Windows 10 KMDF drivers, so these machines seem to be set up correctly. There is clearly no issue communicating with the Turbot, yet this provisioning times out (error 10060, "A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host as failed to respond 192.168.55:50005").

How does one make progress past this? Are there manual steps that can be taken to provision the device for driver development, and if so, what are they?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5302a6a5-9017-4475-25ee-167a9b816d31
* Version Independent ID: f0bff1be-a4fe-3de0-1ea9-7ec77def3b0c
* Content: [Driver Deployment - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/driverdeployment)
* Content Source: [windows-iotcore/learn-about-hardware/DriverDeployment.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/DriverDeployment.md)
* Product: **windows-iot**
* Technology: **iot**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

saraclay, 2019-04-29:
@druske the team is investigating this

saraclay, 2019-05-14:
Sorry for the delay @druske. It seems like you might have an issue with your entered device name. I'm assuming that "192.168.55:50005" looks like an IP address, but it's not. It's missing an octet:  192 . 168 . 55 : 50005.  There should be 4 octets: 192.168. x. 55 : 50005.  That's most likely why VS can't talk to the machine.

druske, 2019-05-15:
That was a typo on my part, sorry. The actual address in the message was 192.168.**143**.55:50005.

Here's another report that may be the same issue:
[https://social.msdn.microsoft.com/Forums/en-US/ace7152e-4f69-49cd-91d4-3be367fde9ca/provisioning-a-computer-for-remote-debugging-in-user-mode-driver-development-error-10060?forum=wdk](https://social.msdn.microsoft.com/Forums/en-US/ace7152e-4f69-49cd-91d4-3be367fde9ca/provisioning-a-computer-for-remote-debugging-in-user-mode-driver-development-error-10060?forum=wdk)

Please understand I am not trying to get tech support for my specific issue here. Rather, I am suggesting that the documentation needs to clarify steps to take when automatic provisioning fails.

That said, I do appreciate your follow-up on this.

saraclay, 2019-05-16:
Thanks @druske, got it. Let me see what we have...

TerryWarwick, 2023-02-07:
Closing due to age.  If issue persists, please submit a new issue.

closed 2023-02-07 by TerryWarwick