MicrosoftDocs/windows-iotcore-docs #1273: Typing mistake in a paragraph

ammusilpa, 2019-05-04:
there is a typing mistake in the above paragraph SET UP A NEW DEVICE.
"Dashboard cannot be used used to setup the Raspberry Pi 3B+."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9df006b9-0eac-67b8-9f24-e712ec436ec8
* Version Independent ID: 491d3781-235d-1ad5-fff4-9185cbc73984
* Content: [Windows 10 IoT Core Dashboard - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/iotdashboard#feedback)
* Content Source: [windows-iotcore/connect-your-device/IoTDashboard.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/IoTDashboard.md)
* Product: **windows-iot**
* Technology: **iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-05-06:
Thanks! I've created a PR for this.

closed 2019-05-06 by saraclay

ammusilpa, 2019-05-11:
ok

ammusilpa, 2019-05-11:
okay