MicrosoftDocs/windows-iotcore-docs #1274: Failed to get insider build

NathanielJS1541, 2019-05-04:
I am using the Windows 10 IoT Core Dashboard to set up a new device. I am registered as a Windows Insider and have signed in with the correct account.

When downloading any Windows Insider Preview image while setting up a new device, I keep getting the _Failed to get insider build_ error. My internet connection appears to be fine, since the image successfully downloads despite the error. Sometimes this error appears when typing in an admin password or selecting a new image.
[image]

I do have Kaspersky Internet Security installed and thought the issue may be down to that. Using Wireshark I found that the Windows 10 IoT Core Dashboard was using port 443, so I stopped my security software from scanning this port but the issue still persists. Should I be concerned about this issue and is there anything I can do to rectify it?

The error also appears when trying to download the standard images that do not require a Windows insider account to use.

I also sometimes get a message saying _Looks like you're not an insider._ but this doesn't happen all the time.
[image]

Cheers.

saraclay, 2019-05-06:
Thank you for the very thorough issue, @NathanielJS1541 - I myself have seen this error but it wasn't until your post that I've seen it reproduced by someone else. I'll file this as a bug. Thanks!

saraclay, 2019-05-09:
Hi @NathanielJS1541, we've made a few changes since you posted this issue. Try and download the newest version of the Dashboard (you'll be prompted to download a new version). If you're still getting an error, please feel free to file a new issue. Thank you!

closed 2019-05-09 by saraclay