MicrosoftDocs/windows-iotcore-docs #1294: Raspberry Pi 3B+ installing appx "failed"

ychen0811, 2019-05-13:
Hello,
I currently have a Qt project that I need to deploy to a Raspberry Pi 3 B+ with Windows 10 IoT Core.
I'm using the Raspberry Pi 3B+ technical preview build and QTcreator. I used the makeappx command to manually make an package to install on my board via Windows device portal.
When I chose to install the appx, I got a error message which said that the OS version is outdated and therefore couldn't install the .appx .  When I repeated the same steps on my Rapsberry Pi 2B with the latest buid, all worked very fine. 
Is there anyway to deploy a c++/Qt project to a R3b+ ? (I failled to deploy C++ projects to the R3b+ with Visual Studio as well)

Thanks

saraclay, 2019-05-13:
Hi @ychen0811, if this is the case, the 3B should probably work for you. Is using 3B+ a hard requirement?

ychen0811, 2019-05-14:
Thanks for the reply. Indeed, Raspberry 3B+ is a must have for my project.

saraclay, 2019-05-14:
Not to be a bother, but why is that? While we will have another release of 3B+ at some point, we do not have a ETA right now for it. If this project is urgent, I would recommend using the 3B.

ychen0811, 2019-05-15:
Actually, for an industrial usage, I do prefer a longer lifespan for the platform and therefore, choosing the latest version of the Rasperry PI.

saraclay, 2019-05-15:
If that's the only reason, Raspberry 3B has actually been commercialized a number of times with IoT Core. But I will leave it to you if you must use the 3B+.

ychen0811, 2019-05-16:
Thank you for the information @saraclay . I've managed today to install a UWP C++ App on the RP3b+ thanks to the package creator wizard available on Visual Studio 2017. I've just needed to adjust the OS version before creating the package.

saraclay, 2019-05-16:
That's great to hear!

closed 2019-05-16 by saraclay