MicrosoftDocs/windows-iotcore-docs #1360: App Installation Instructions are badly out of date

scout208, 2019-06-03:
The instructions on this page https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller haven't been updated since 2017 and many things have changed since then so some of the instructions and images are no longer valid. Additionally, installing apps the ways described in this page doesn't allow you to start them if the package is a .msix file which is the only supported package version in the latest version of Visual Studio so the instructions are now obsolete.

saraclay, 2019-06-03:
Thank you @scout208 - you're absolutely right. I will raise this internally and ask this to be prioritized.

saraclay, 2019-06-06:
These have been updated.

closed 2019-06-06 by saraclay