MicrosoftDocs/windows-iotcore-docs #1442: RPi 4B support

RyanMorash, 2019-06-30:
Are there any plans to add support for the newly released Raspberry Pi 4 Model B using the BCM2711 SoC. A number of significant hardware changes have been made including:

- moving the the boot loader from the SD card to EEPROM
- moving the USB (now USB 3.0) to PCIe bus
- "the Gigabit Ethernet driver is completely different to previous models"

bormm, 2019-07-05:
You should not expect anything. Even the Pi 3B+ is not supported. Just take a look at my issue #519 and the other ones: #1172 #1082 #962.

ericgrantholland, 2019-08-30:
MS is falling too far behind, very sad. The market for IoT/embedded systems is huge and they have Azure which has been performing well every quarter, yet they are falling behind on IoT which should integrate directly with Azure services?!? 

As to why use windows for IoT? well, UWP is a better UI framework than anything else I've seen. On top of that, it's what I'm most familiar with.

TerryWarwick, 2023-02-07:
Closing due to age. If issue persists, please submit a new issue.

closed 2023-02-07 by TerryWarwick