MicrosoftDocs/windows-iotcore-docs #1451: How to enable on-screen keyboard on Raspberry Pi 3 B+

memory125, 2019-07-10:
I've installed Windows 10 IoT Core image 17661 to RPi 3B+, and deployed UWP app to it. But when we want to input something, the `TextBox` can't invoke the on-screen keyboard automatically, we configured the setting according to [on-screen keyboard configuration](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/onscreenkeyboard#non-obscured-text-input-fields), but the issue is existed as well.

memory125, 2019-07-10:
And we just validated on Windows 10 PC, also can't show the on-screen keyboard even this device is in tablet mode.

memory125, 2019-07-11:
Resolved.

closed 2019-07-11 by memory125

coco-1998-2, 2019-07-22:
We could find the checkbox "Show the on-screen keyboard when text input controls have focus" in "Device Setting" part of Windows 10 Iot Core management portal.

memory125, 2019-07-23:
> We could find the checkbox "Show the on-screen keyboard when text input controls have focus" in "Device Setting" part of Windows 10 Iot Core management portal.

Exactly