MicrosoftDocs/windows-iotcore-docs #1455: Update "Connect your app to the cloud" page for VS2019

zivkan, 2019-07-21:
The page `windows-iotcore/connect-to-cloud/ConnectAppToCloud.md` basically says to use a connected service extension in VS2017. However, that extension was never ported to VS2019.

saraclay, 2019-07-25:
Have you tried it with 2019? I believe it still works. Let me know if you're running into issues.

zivkan, 2019-07-26:
Yes, I've tried VS2019, but when I go to add a new connected service, Azure IoT Hub is not listed. I've installed the "Azure development" workload, and I don't see anything that looks relevant in the individual components list.

saraclay, 2019-09-23:
Please post this on [our forums](https://social.msdn.microsoft.com/Forums/vstudio/en-US/home?forum=WindowsIoT). Our GitHub issues are for documentation related inquiries only. Thank you.

closed 2019-09-23 by saraclay