MicrosoftDocs/windows-iotcore-docs #1564: Trouble Enabling BitLocker

Zarshua, 2020-01-30:
I'm having trouble verifying that these generated test packages are working. I started out trying to enable all of the packages at once, but to reduce the number of variables I decided to see if I could just get BitLocker working.

I can generate all the packages and follow the instructions past the 'applyupdate -commit' command (after staging the bitlocker package alone). After that, my device will reboot and display the two gears/status bar... then, the device boots up normally and sits at the default foreground app.

At this point, I start a PSSession with the device and run 'start /wait sectask.exe -waitencryptcomplete:1' but nothing is returned.

I uncommented the rest of the 'setup.bitlocker.cmd' script, and can see from the 'bitlockerstatus.log' file that 'SecTask.exe' is timing out... but there is no reason provided.

Is there a way to get more information about why 'SecTask.exe' is timing out?

Thanks!

imingc, 2020-01-31:
Bitlocker requires SecureBoot to be enabled, so if you're testing individual packages, the first one you should try is to enable SecureBoot.  Note that enabling SecureBoot will require that all drivers are correctly signed with driver signing certificates.  See #8 on 
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/set-up-your-pc-to-customize-iot-core

Zarshua, 2020-02-03:
Thanks Jimmy, I'll see where I can get with this information.

Zarshua, 2020-02-04:
I'd like to be able to just demonstrate that these features work on our hardware -- #8 in https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/set-up-your-pc-to-customize-iot-core appears to refer to a 'retail' image.

I'm trying to build a 'test' image first, so I generated certificates based off of this article, #4: https://docs.microsoft.com/en-us/windows/iot-core/secure-your-device/securebootandbitlocker#turnkey-security-on-iot-core

According to this document: https://github.com/ms-iot/iot-adk-addonkit/tree/master/Tools#adding-security-packages, I've added the PK, UEFISB, and DRA certificates to my IoTWorkspace.xml file (I'm skipping DeviceGuard for now), built/added the packages, modified 'oemcustomization.cmd' and then built a 'test' image.

When I deploy this new image on my device, I can see that the C:\IoTSec folder is being created, and I see the cmd files that set registry bits and schedule tasks were created... so I know the packages I built and embedded in IoT Core are being run... but I can't see where they're failing, or what I need to provide for them to work.

Zarshua, 2020-02-17:
Opening another issue to hopefully get a response.

closed 2020-02-17 by Zarshua