MicrosoftDocs/windows-iotcore-docs #1588: Windows Hardware Manufacturing documentation is not on GitHub

rossmpersonal, 2020-06-07:
The Windows Hardware documentation, including the Start building Windows 10 devices documentation, the Windows Device design documentation, the Windows Customization documentation, the Windows Deployment documentation, the Windows Manufacturing documentation, the Windows Hardware testing documentation, the Windows Hardware Performance documentation, the Windows ADK documentation, including but not limited to https://docs.microsoft.com/en-us/windows-hardware/get-started/ , https://docs.microsoft.com/en-us/windows-hardware/design/ , https://docs.microsoft.com/en-us/windows-hardware/customize/ , https://docs.microsoft.com/en-us/windows-hardware/test/ , and https://docs.microsoft.com/en-us/windows-hardware/manufacture/ , is not on the MicrosoftDocs GitHub. It needs to be added.

It addition most of the documentation desperately needs to be update.

I realize this might not be best place to post this issue but I could not find a better place.

EliotSeattle, 2020-06-12:
Hi @rossmpersonal 
Thanks for the feedback. I agree that discovery of those content areas is problematic at best. I'll be leading an effort to fix this.
I'm going to close this issue and have already opened [issue #2328](https://github.com/MicrosoftDocs/windows-driver-docs/issues/2328) in a repo that is more closely connected to the hardware content and which I also manage.
Thanks,
Eliot

EliotSeattle, 2020-06-12:
@TerryWarwick can you please close this issue? I'll cover this request in issue #2328, tracked in the driver repo.
Thanks,
Eliot

rossmpersonal, 2020-06-12:
Thank you so much. I would like to clarify that while this probably more related to Windows Drivers than Windows Not Core, the manufacturing I am referring to, which honestly is a complete mess, is really unrelated to Windows Drivers. It is really its own beast and desperately needs to be cleaned up, updated, moved on GitHub, and possibly rewritten. Thank you so much taking this on.

EliotSeattle, 2020-06-16:
Hi @rossmpersonal 
Go ahead and take a look at  the content now. We've piped it all together and tried to make the navigational experience smooth. Did we hit the mark?
Updating the content is more complex and will take more time.

rossmpersonal, 2020-06-16:
I am really not good with testing or evaluating the navigational experience. I am really not a UI/UX person. I did notice some improvements though.

Some navigation issues remain. The hierarchy of the content is confusing and inconsistent. There is no page that is really the top of the hierarchy or that visibly serves that purpose. It is unclear if there is even a path to navigate to that area of documentation from docs.microsoft.com without going to other documentation that randomly links to it and discovering that it exists by accident.

The order of items in the left navigation outline does not make much sense. I would think that it would go in a logical order like:
What's new
Getting started
Design
Customize
Test
Manufacturing
Deploy
Service

The current order does not seem to be logical. Furthermore, the order varies from page to page making it even more confusing.

To make matters worse, if you navigate down into the hierarchy and then navigate up in the hierarchy, there are places where you do not come back to the page you navigated "down" from making it easy to "get lost" and makes it unclear how things are organized.

Yes, I see improvement but I see a lot of work to be done.

I am sorry for being so picky.

Thank you.

EliotSeattle, 2020-06-16:
Hi @rossmpersonal 

You're not picky at all. I really appreciate your feedback. They're all quite insightful. I'll consider how to get to these improvements and weigh the scheduling and priority against the other work we're obligated to complete. 

If you think of any other changes you'd like to see, please either create a new issue or add them to this one. I'll close this one, as the original request of connecting repos and sites is complete, and I've already created tasks in our internal backlog for the ones in your latest comment.

Thanks so much for your time and input.
Eliot

rossmpersonal, 2020-06-16:
One other thing I forgot to mention: There is documentation for desktop manufacturing and iot core manufacturing but there is no apparent documentation for server manufacturing. Yes, I would expect server manufacturing to be a superset of desktop manufacturing but I would still think it should be documented. Thank you.

tedhudek, 2020-06-18:
Hi @rossmpersonal, thanks for the feedback. Sadly there is no easy way to give these docs a top level landing page. I did go ahead and rearrange the TOC nodes as you can now see at https://docs.microsoft.com/en-us/windows-hardware/get-started/.  I also added a nav card called "Build devices" to https://docs.microsoft.com/en-us/windows-hardware/drivers/, which in turn you can get to from https://docs.microsoft.com/en-us/windows.  Hope this helps!

rossmpersonal, 2020-06-18:
This is certainly an improvement. I like the links from the hardware developer docs page. On the TOC, I would certainly think that test for performance and compatibility would come before service. I cannot decide whether test for performance and compatibility should come before or after manufacture. That would probably a good question for a windows manufacturing SME.

tedhudek, 2020-06-18:
Thanks- I did actually ask a colleague who is my closest contact for this area, and the new order is his recommendation.

closed 2023-02-07 by TerryWarwick