MicrosoftDocs/windows-iotcore-docs #1611: About Windows IOT core for INTEL JOULE. Need help.

mopplayer, 2020-08-31:
Hi,

Is there a ISO for INTEL JOULE.
How to download the latest ISO for **"Windows10_InsiderPreview_IoTCore_IntelJoule_ARM32ARM64_en-us_16267.iso".**

Thanks.

chsha, 2020-09-04:
Intel discontinued the Joule product line several yeas ago - https://ark.intel.com/content/www/us/en/ark/products/series/96418/intel-joule-products.html. There is no updated ISO with the latest IoT Core OS. You may try to contact Intel to see if they have drivers which you can use to manually build a system image.

mopplayer, 2020-09-04:
Hi,

I know the product is end of life.
But I just want to try Windows IOT core for JOULE.
Is there the latest ISO?

Thanks.

chsha, 2020-09-04:
No. We no longer produce an ISO for Joule with the latest IoT Core OS.

mopplayer, 2020-09-04:
Hi,

I mean 16267 version ISO.

Thanks.

closed 2023-02-07 by TerryWarwick