MicrosoftDocs/windows-iotcore-docs #1711: Soft Real-Time Setup description missing

kdotdk, 2022-12-15:
Hi

I am looking for a description on how to configure a Windows IoT Core for Soft Real-Time.
I have found a description for [IoT Enterprise](https://learn.microsoft.com/en-us/windows/iot/iot-enterprise/soft-real-time/soft-real-time-application), but not for IoT Core.
Basically a page with IoT Core Real-Time information is missing.
(Or will the functionality first be available in next release?)

Regards

TerryWarwick, 2022-12-15:
kdotdk, there are no plans to support Soft Real-Time on IoT Core. 

I should also point out that IoT Core alone is out of support. To continue receiving servicing updates on IoT Core the device must be enrolled in IoT Core Services.

kdotdk, 2022-12-16:
> no plans to support Soft Real-Time on IoT Core

Question:
Does **_IoT Core Service_** support ProcessorAffinity and setting threads to Realtime priority?

> IoT Core alone is out of support.

Ok

Follow-up questions:
- Does this imply this [documentation](https://learn.microsoft.com/en-us/windows/iot-core/windows-iot-core) is only for _**IoT Core alone**_ and not valid for **_IoT Core Service_**?
- When manufacturing a new device under **_IoT Core Service_** following the [manufacture guide](https://learn.microsoft.com/en-us/windows-hardware/manufacture/iot/iot-core-manufacturing-guide?view=windows-11) will I end up with a Windows device with end of life support Jan 9, 2029 and not today+10year?
(This [page](https://learn.microsoft.com/en-us/lifecycle/products/windows-10-iot-core-services) states **_IoT Core Service_** is supported via [Windows 10 IoT 2019 LTSC](https://learn.microsoft.com/en-us/lifecycle/products/windows-10-iot-enterprise-ltsc-2019) with EOL Jan 9, 2029)

TerryWarwick, 2023-02-07:
Together IoT Core and IoT Core Services provide you with the Windows 10 IoT Core LTSC lifecycle with End of Support being Jan 9, 2018.  

- [IoT Core Documentation](https://learn.microsoft.com/en-us/windows/iot-core/windows-iot-core)
- [IoT Core Services Documentation](https://learn.microsoft.com/windows-hardware/manufacture/iot/iotcoreservicesoverview)

IoT Core Services does not change the capabilities of the IoT Core, it provides extended support for servicing updates.

closed 2023-02-07 by TerryWarwick